landonr/lilygo-tdisplays3-esphome #5: Compile errors

muzzy996, 2022-12-28:
Noob here but hoping for some help.  What exactly does "ensuring to change the s3.yaml api" to one's own credentials?  My secrets.yaml should be fine as it has my correct wifi SSID and password info but I didn't know what to change in the s3.yaml file so I left it alone.  

In any event, when trying to complete the manual download I'm getting a bunch of errors and eventual failure.  Log attached. 

[logs_s3_compile.txt](https://github.com/landonr/lilygo-tdisplays3-esphome/files/10311772/logs_s3_compile.txt)

patfelst, 2023-01-06:
I think that means to change your wifi credentials, which you've done. It possibly means to change the captive portal "ap" details but that can be left as-is. It sounds like you have some other problem, probably yaml formatting. I haven't actually tried this code yet, but if you post your yaml, we might be able to spot an error. When you post, use code formatting ```yaml  then paste your code, then close also with three backticks.

To be honest this is pretty advanced if you're new to ESPHome.

guillempages, 2023-06-25:
@muzzy996 from your error logs, it looks like you have some html embedded in your yaml config; which does not make sense IMHO.
Did you actually check the contents of your yaml file to see what's inside?

closed 2023-11-14 by landonr