landonr/lilygo-tdisplays3-esphome #6: OTA Flash dosent work.

eisaheinig, 2023-01-10:
My Tdsiplays3 is up and running, displaying the time and date.
I followed the Tutorial up until the part where we move the "TFT_eSPI tft = TFT_eSPI();" to the Public part.
Then Re-Compiled and tried to flash via the OTA this did not work, do we always need to connect it via usb to flash it?

```
INFO Reading configuration /config/esphome/s3.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing s3 (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32 @ 5.0.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- SPI @ 2.0.0
|-- FS @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- TFT_eSPI @ 2.4.71+sha.986ca0a
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
Compiling /data/s3/.pioenvs/s3/src/main.cpp.o
In file included from /data/s3/.piolibdeps/s3/tdisplays3/TFT_eSPI.h:68,
                 from src/tft_espi_display.h:2,
                 from src/main.cpp:28:
/data/s3/.piolibdeps/s3/tdisplays3/Processors/TFT_eSPI_ESP32_S3.h:11:3: warning: #warning >>>>------>> DMA is not supported on the ESP32 S3 (possible future update) [-Wcpp]
  #warning >>>>------>> DMA is not supported on the ESP32 S3 (possible future update)
   ^~~~~~~
In file included from src/tft_espi_display.h:2,
                 from src/main.cpp:28:
/data/s3/.piolibdeps/s3/tdisplays3/TFT_eSPI.h:882:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
        ^~~~~~~
Linking /data/s3/.pioenvs/s3/firmware.elf
RAM:   [=         ]  12.1% (used 39664 bytes from 327680 bytes)
Flash: [====      ]  41.1% (used 755009 bytes from 1835008 bytes)
Building /data/s3/.pioenvs/s3/firmware.bin
Creating esp32s3 image...
Successfully created esp32s3 image.
esp32_create_combined_bin(["/data/s3/.pioenvs/s3/firmware.bin"], ["/data/s3/.pioenvs/s3/firmware.elf"])
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=8MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Wrote 0xc86c0 bytes to file /data/s3/.pioenvs/s3/firmware-factory.bin, ready to flash to offset 0x0
========================= [SUCCESS] Took 34.16 seconds =========================
INFO Successfully compiled program.
INFO Resolving IP address of s3.local
INFO  -> ********
INFO Uploading /data/s3/.pioenvs/s3/firmware.bin (755392 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
ERROR Error Update end: Error: Finishing update failed. See the MQTT/USB logs for more information.

```

eisaheinig, 2023-01-10:
I connected the esp to the Server, and flashing again was successfully, i was hoping we can use EspHome OTA for flashing further the device. Or am i doing it Wrong.
k.R

fisheradam, 2023-01-10:
Unfortunately I have also been unable to flash via OTA but no issues via USB.

> I connected the esp to the Server, and flashing again was successfully, i was hoping we can use EspHome OTA for flashing further the device. Or am i doing it Wrong. k.R

eisaheinig, 2023-01-10:
okay than this can be closed :)

eisaheinig, 2023-01-10:
lets close this than, thanks for reply @fisheradam