landonr/lilygo-tdisplays3-esphome #7: Black screen

eisaheinig, 2023-01-10:
Anyone has an Idea my screen show after last Compile & install a blackscreen:

`INFO Reading configuration /config/esphome/s3.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'Europe/Berlin'
INFO Generating C++ source...
INFO Compiling app...
Processing s3 (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32 @ 5.0.0)
--------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/landonr/lilygo-tdisplays3-esphome.git
INFO Installing git+https://github.com/landonr/lilygo-tdisplays3-esphome.git
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-la91mwkk'...
Library Manager: tdisplays3@2.4.71+sha.986ca0a has been installed!
INFO tdisplays3@2.4.71+sha.986ca0a has been installed!
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
Dependency Graph
|-- SPI @ 2.0.0
|-- FS @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- TFT_eSPI @ 2.4.71+sha.986ca0a
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
Compiling /data/s3/.pioenvs/s3/src/esphome/components/api/api_connection.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/api/api_server.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/api/list_entities.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/api/proto.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/api/user_services.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/esp32/core.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/esp32/gpio.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/esp32/preferences.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/gpio/switch/gpio_switch.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/homeassistant/text_sensor/homeassistant_text_sensor.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/logger/logger.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/md5/md5.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/mdns/mdns_rp2040.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/network/util.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/ota/ota_backend_arduino_rp2040.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/socket/socket.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/switch/automation.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/switch/switch.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/text_sensor/filter.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/text_sensor/text_sensor.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/time/automation.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/time/real_time_clock.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/wifi/wifi_component.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/wifi/wifi_component_esp32_arduino.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/wifi/wifi_component_esp_idf.cpp.o
Compiling /data/s3/.pioenvs/s3/src/esphome/components/wifi/wifi_component_pico_w.cpp.o
[... 177 lines cut ...]
Configuring flash size...
Auto-detected Flash size: 16MB
Flash will be erased from 0x00010000 to 0x000c8fff...
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Compressed 757680 bytes to 492858...
Wrote 757680 bytes (492858 compressed) at 0x00010000 in 5.2 seconds (effective 1160.4 kbit/s)...
Hash of data verified.
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=16MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Compressed 13824 bytes to 9619...
Wrote 13824 bytes (9619 compressed) at 0x00000000 in 0.2 seconds (effective 444.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.1 seconds (effective 386.6 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 568.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyACM0 with baud rate 115200
`

landonr, 2023-01-12:
were you able to get this working?

eisaheinig, 2023-01-14:
Yes, I got it to work, I am still struggling to display more home assistant content.

For now, it just displays a string and the IP Address.
I wish there would be a straight forward library/add-on like the dashboards/cards in Home Assistant.
cya

closed 2023-06-27 by landonr