landonr/lilygo-tdisplays3-esphome #8: Compile warnings and notes

rperciaccante, 2023-01-13:
I am trying to compile the s3.yaml example, and I am getting the following feedback:

```
C:\Users\bob\esphome\projects\chickencoupe>esphome compile compiletest.yaml
INFO Reading configuration compiletest.yaml...
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
WARNING The selected Arduino framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.
INFO Detected timezone 'America/New_York'
INFO Generating C++ source...
INFO Compiling app...
Processing s3 (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32 @ 5.0.0)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing git+https://github.com/landonr/lilygo-tdisplays3-esphome.git
INFO Installing git+https://github.com/landonr/lilygo-tdisplays3-esphome.git
git version 2.34.1.windows.1
Cloning into 'C:\Users\bob\.platformio\.cache\tmp\pkg-installing-u68ws9zy'...
remote: Enumerating objects: 687, done.
remote: Counting objects: 100% (687/687), done.
remote: Compressing objects: 100% (461/461), done.
remote: Total 687 (delta 149), reused 677 (delta 149), pack-reused 0
Receiving objects: 100% (687/687), 3.73 MiB | 13.58 MiB/s, done.
Resolving deltas: 100% (149/149), done.
Updating files: 100% (569/569), done.
Library Manager: tdisplays3@2.4.71+sha.986ca0a has been installed!
←[32mINFO ←[32mtdisplays3@2.4.71+sha.986ca0a has been installed!←[0m←[0m
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (5.0.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20003.220626 (2.0.3)
 - tool-esptoolpy @ 1.30300.0 (3.3.0)
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch3
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 33 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 2.0.0
|-- FS @ 2.0.0
|-- SPIFFS @ 2.0.0
|-- TFT_eSPI @ 2.4.71+sha.986ca0a
|-- WiFi @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- Update @ 2.0.0
Building in release mode
Compiling .pioenvs\s3\src\esphome\components\api\api_connection.cpp.o
Compiling .pioenvs\s3\src\esphome\components\api\api_frame_helper.cpp.o
Compiling .pioenvs\s3\src\esphome\components\api\api_pb2.cpp.o
Compiling .pioenvs\s3\src\esphome\components\api\api_pb2_service.cpp.o
Compiling .pioenvs\s3\src\esphome\components\api\api_server.cpp.o
Compiling .pioenvs\s3\src\esphome\components\api\list_entities.cpp.o
Compiling .pioenvs\s3\src\esphome\components\api\proto.cpp.o
Compiling .pioenvs\s3\src\esphome\components\api\subscribe_state.cpp.o
Compiling .pioenvs\s3\src\esphome\components\api\user_services.cpp.o
Compiling .pioenvs\s3\src\esphome\components\esp32\core.cpp.o
Compiling .pioenvs\s3\src\esphome\components\esp32\gpio.cpp.o
Compiling .pioenvs\s3\src\esphome\components\esp32\preferences.cpp.o
Compiling .pioenvs\s3\src\esphome\components\gpio\switch\gpio_switch.cpp.o
Compiling .pioenvs\s3\src\esphome\components\homeassistant\time\homeassistant_time.cpp.o
Compiling .pioenvs\s3\src\esphome\components\logger\logger.cpp.o
Compiling .pioenvs\s3\src\esphome\components\md5\md5.cpp.o
Compiling .pioenvs\s3\src\esphome\components\mdns\mdns_component.cpp.o
Compiling .pioenvs\s3\src\esphome\components\mdns\mdns_esp32_arduino.cpp.o
Compiling .pioenvs\s3\src\esphome\components\mdns\mdns_esp8266.cpp.o
Compiling .pioenvs\s3\src\esphome\components\mdns\mdns_esp_idf.cpp.o
Compiling .pioenvs\s3\src\esphome\components\mdns\mdns_rp2040.cpp.o
Compiling .pioenvs\s3\src\esphome\components\network\util.cpp.o
[... 167 lines cut ...]
Compiling .pioenvs\s3\FrameworkArduino\stdlib_noniso.c.o
Compiling .pioenvs\s3\FrameworkArduino\wiring_pulse.c.o
Compiling .pioenvs\s3\FrameworkArduino\wiring_shift.c.o
Archiving .pioenvs\s3\libFrameworkArduino.a
Linking .pioenvs\s3\firmware.elf
Retrieving maximum program size .pioenvs\s3\firmware.elf
Checking size .pioenvs\s3\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.1% (used 39664 bytes from 327680 bytes)
Flash: [====      ]  41.1% (used 755025 bytes from 1835008 bytes)
Building .pioenvs\s3\firmware.bin
esptool.py v3.3
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
esp32_create_combined_bin([".pioenvs\s3\firmware.bin"], [".pioenvs\s3\firmware.elf"])
esptool.py v4.4
Warning: Image file at 0x0 is protected with a hash checksum, so not changing the flash size setting. Use the --flash_size=keep option instead of --flash_size=8MB in order to remove this warning, or use the --dont-append-digest option for the elf2image command in order to generate an image file without a hash checksum
Wrote 0xc86d0 bytes to file C:\Users\bob\esphome\projects\chickencoupe\.esphome\build\s3\.pioenvs\s3/firmware-factory.bin, ready to flash to offset 0x0
================================================================================= [SUCCESS] Took 266.10 seconds =================================================================================
←[32mINFO Successfully compiled program.←[0m

C:\Users\bob\esphome\projects\chickencoupe>
```
Is there anything that I can do to eliminate these?

landonr, 2023-01-23:
i wouldn't put much effort into fixing these. this repo is kinda a hack to get the display working with esphome until someone smart enough writes drivers for the standard display library

guillempages, 2023-07-04:
#31 should hopefully fix this

closed 2023-07-05 by landonr