landonr/lilygo-tdisplays3-esphome #9: Help with showing sensor data from homeassitant

dborup, 2023-01-19:
Hi

Can anybody help me showing sensor data instead of text?

landonr, 2023-01-23:
it should be something like 
```
std::string displayString = atof(id(sensor_id).state());
displayControllerComponent->drawString(displayString);
```

dborup, 2023-01-23:
> it should be something like
> 
> ```
> std::string displayString = atof(id(sensor_id).state());
> displayControllerComponent->drawString(displayString);
> ```

I am not sure I get it.

I have this:
sensor:
  - platform: homeassistant
    id: inside_temperature
    entity_id: sensor.badevaerelse_sensor_temperature
    internal: true
interval:
  - interval: 1s
    then:
    - lambda: |-
        std::string displayString = atof(id(inside_temperature).state());
        displayControllerComponent->drawString(displayString);

and get this error 
x/config/esphome/s3.yaml: In lambda function:
/config/esphome/s3.yaml:47:66: error: expression cannot be used as a function
         std::string displayString = atof(id(inside_temperature).state());

landonr, 2023-01-24:
sorry i tried coding without compiling and this is what happens
this worked for me
```
  std::string displayString = to_string(id(inside_temperature).state);
```

closed 2023-07-05 by landonr