landonr/lilygo-tdisplays3-esphome #11: Enable usage as an external component

guillempages, 2023-02-26:
It would be nice to be able to use this component as a standard display, supporting all the configuration options and lambdas that other displays support.

hapklaar, 2023-03-04:
This works great! I have my energymeter now working on this module using your code. Only issue I found thus far is the display seems to redraw the screen a bit slow, so that it flashes every update interval. My other module (standard T-Display) doesn't do this. Made a little vid to illustrate. Wondering if anything can be done about this?

https://user-images.githubusercontent.com/7703320/222916441-6454d642-f3c8-4a06-b668-aecdc7efba41.mp4

brooksbennett, 2023-03-04:
Same issue for me.

guillempages, 2023-03-04:
I did notice the flickering when implementing it, but since it is the first LCD Display I use with esphome I thought it was normal. I'm used to e-ink displays, and such a "short" refresh would be a dream there 😂

Your video proves me wrong. Which component are you using @hapklaar? I can try looking at that coffee and see if I spot some difference.

In general, I did not develop the component; i just limited it with the standard display instead of a completely custom component. 😉

hapklaar, 2023-03-04:
I'm not sure what you mean with component?

guillempages, 2023-03-04:
I mean which "platform" are you using on the "display" configuration

hapklaar, 2023-03-04:
ah for the other module:

display:
  - platform: st7789v
    model: TTGO TDisplay 135x240

guillempages, 2023-03-04:
I've looked at the st7789v code, and as far as I can see, the latter one is using some sort ouf double-buffering, to make updates smoother.
This would be the right thing to do here as well (for the t-display-s3). It *is* using an st7789v controller as well far as I know, so the optimal thing to do would be to add support for it to the core st7789v component (as an additional model).
That would be way beyond what my PR is about; it is only allowing the work done by @landonr to be used directly as a standard esphome display, instead of having to tweak and use the "TFT_eSPI" class directly. 
I agree it is not perfect, but hey, at least it's showing something ;-)

Sorry that I don't have better news :-(

hapklaar, 2023-03-05:
That's ok, thanks for having a look! Havinf it working like this is already a big step forward. Hopefully we will get some native support for this module in esphome in the near future.

bradmck, 2023-03-09:
So I'm _definitely_ not an expert on this stuff but I also experienced the flashing on draw/refresh issue.  I'm still struggling a bit but wanted to mention that Sprites from TFT_eSPI seemed to work around that.  Instead of drawing to the screen one element at a time you draw to the Sprite (which can contain other Sprites) and then push that Sprite to the display all at once.  Mentioning just in case it helps someone else since it made a big difference for me.

And a question if I may.....  I wasn't going to post a new topic but while I'm posting I'll see if any of you know this.  The main TFT_eSPI library now has a UserSetup file (User_Setups/Setup206_LilyGo_T_Display_S3.h) for the T-Display-S3.  Does anyone know how to make it so we can use the main library instead of custom patched versions for this display?  I **really** appreciate the work done on this repository by the way.  I'd have been totally lost had I not found this.  I can make this display do exactly what I want using the Arduino IDE but would really like to use ESPHome.  Is it just a matter of ESPHome updating the TFT_eSPI library to be more current?

**PS on Sprites**:  I just added some Sprite objects to the tft_espi_display.h file I'm calling and can use those objects in Lambdas:
`	TFT_eSprite spr = TFT_eSprite(&tft); // Sprite object
	TFT_eSprite sprBackground = TFT_eSprite(&tft); // Sprite object
	TFT_eSprite sprRSSIGauge = TFT_eSprite(&tft); // Sprite object`

hapklaar, 2023-03-10:
couldn't wait so changed the code to get your latest changes :)

```
external_components:
  - source: github://guillempages/lilygo-tdisplays3-esphome@use_sprites
    components: [tdisplays3]
```

works great, no more flickering! Thanks @bradmck and @guillempages

guillempages, 2023-03-11:
@bradmck I've looked at the st7789v code from ESPHome, and as far as I can see it is not using the TFT_eSPI library at all, but implementing the communication using directly SPI. So it's not as easy as it sounds, to use the newer version of this library.
I'll leave the implementation for upstream to the experts ;-)

bradmck, 2023-03-11:
Thanks for looking into the ESPHome code @guillempages !  And I'm glad my sprites suggestion was helpful.  To me the ideal for sprites would be to able to easily create new sprite objects in lambdas.....  They're very useful for being able to update individual elements on screen, can have transparent background, and sprites can be pushed into other sprites.  So you build and fill your background sprite, then build other sprites and push into the background sprite, then draw the background sprite to the screen all at once.  Again, I am definitely not an authority on any of this though. 

I'll mention something else in case it's helpful to you since you obviously know how to write code.....  If I use a different version of the TFT_eSPI library I can actually setup the backlight using ledc so it's dimmable.  That doesn't work with the library included here.  I realize I changed topics but seemed worth mentioning to you.

bradmck, 2023-03-12:
@guillempages @hapklaar @landonr Check the issue I opened up #15 --- I wanted to be able to use the stock TFT_eSPI library and I seem to have that working.  Just looking to share.

guillempages, 2023-03-12:
Hi @bradmck I've briefly tried to use the stock library in the "external_components" variant, and tried putting the defines you are using into the "init" code of the component, to avoid having to add them to the yaml file, but did not succeed.

It would have been nice, but in any case, we seem to have different goals: You want to use the stock library, to be able to set the backlight dimming and maybe use sprites, and I want to have the component as easy to use as possible (i.e. by just adding the external_component configuration and treating it like a stock ESPHome display) :-)
AFAIK other ESPHome displays do not have sprite support, so adding this in the external_component variant doesn't look like the way to go to me.

Fortunately, currently both approaches can be used with this repository :-)

That all said, it's still @landonr 's repository after all, so he should decide which direction he wants to go (or maybe both).

bradmck, 2023-03-12:
Understood @guillempages - like I said just sharing what I've learned.  My goal is actually to be able to use both methods - I want the simplicity of the display component but I'd also like to be able to draw more complicated gauges and stuff.  I also do want to stick with a stock library vs. patched.  I do definitely want the dimmable backlight, but that should be doable either way.  If the BL pins are enabled in the library (or passed as build args) then dimming no longer works.

I'm going to keep working on this and if I make more progress I'll update this in case it's of interest to anyone.  Really appreciate the work you guys have put into this!

guillempages, 2023-03-13:
The display can now be used as an external_component with minimal configuration; closing issue.

closed 2023-03-13 by guillempages