landonr/lilygo-tdisplays3-esphome #12: Very promising work just sending kudos :)

njbuch, 2023-02-27:
Hi there Landon and Adam

Just a heads-up from here, awaiting your work, so I can get started on deploying some TTGO S3 display devices on my setup.

It really looks promsing, but still over the top in terms of me getting it to work.

Thanks ;)

guillempages, 2023-03-11:
@njbuch you might want to try using #14 as an external component; with it, you should be able to use this display as any other ESPHome display and follow other examples that you might find elsewhere to display meaningful data.

njbuch, 2023-04-01:
I tried that, but never got it to work, did you progress on this?

guillempages, 2023-04-01:
That PR was merged, so just using landonr's repository for the external components should work.
There is an "example.yaml" file there that you can use as a basis for your configuration, and iterate on that.

njbuch, 2023-04-03:
Hmm, I tried again pulling the latest version. Ending in an infinite loop of re-starting, the device (usb-alert sound in windows keeps repeating while its attached.)

Here is what I do.
1. Pulling latest version of esphome
2. Pulling the latest version of  https://github.com/landonr/lilygo-tdisplays3-esphome
3. Copying the example.yaml and the tdisplays3 folder to my esphome dir.
4. Doing a "esphome run example.yaml , here is output:

`Linking .pioenvs\s3\firmware.elf
Retrieving maximum program size .pioenvs\s3\firmware.elf
Checking size .pioenvs\s3\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.3% (used 40312 bytes from 327680 bytes)
Flash: [====      ]  44.3% (used 813813 bytes from 1835008 bytes)
Building .pioenvs\s3\firmware.bin
esptool.py v4.2.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
esp32_create_combined_bin([".pioenvs\s3\firmware.bin"], [".pioenvs\s3\firmware.elf"])
esptool.py v4.5.1
Wrote 0xd6c80 bytes to file C:\Code\Home\ESPHomeInfra\.esphome\build\s3\.pioenvs\s3/firmware-factory.bin, ready to flash to offset 0x0` 

5. Going to https://web.esphome.io/ and installing for first use. Succeeding without hickups
6. Connecting again https://web.esphome.io/ and now "install" choosing the firmware from step 4. Waiting for erase and success.
7. Unounting and then disconnect from usb, and re-inserting.
8. Goes into endless loop

Where can I investigate further?

landonr, 2023-04-04:
can you share your usb logs and your yaml

njbuch, 2023-04-04:
The yaml is an exact copy of the example.yaml, only changed the secrets.yaml to match my network.
There is no output from the device, it just keeps rebooting.

landonr, 2023-04-04:
you can see the usb logs at https://web.esphome.io/ if you connect and then hit logs, then press reboot. it seems like its boot looping but there may be something helpful there

njbuch, 2023-04-04:
The only thing I am able to produce is:

``
Terminal disconnected: NetworkError: The device has been lost.
``

Please note to upload the firmware, I choose the USB/JTAG port.
[image]

To me, it seems that some more fundamental boot firmware is missing, is there a prep-step before the esphome "Prepare for first use" can run?

There are 3 bin files in the build folder:
[image]

Would it make sense to upload one of these other than the firmware.bin?

guillempages, 2023-04-04:
I think i had the same issue at the beginning; i could not claim the device.
It started working after flashing the "firmware-factory.bin" file

njbuch, 2023-04-04:
I have now uploaded the firmware-factory.bin file giving this output:
``
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))
Saved PC:0x40378c9d
waiting for download
``

Uploading the generated firmware.bin still ends in bootloop after that.

Also tried 1. upload firmware-factory.bin, 2. "Prepare for first use", 3. Upload firmware.bin. But no change.

patfelst, 2023-04-04:
Hi @njbuch have you tried not doing steps 1 - 3, but let ESPHome pull directly from github when you install? i.e. do a clean files first.

> Here is what I do.
> 
> 1. Pulling latest version of esphome
> 2. Pulling the latest version of  https://github.com/landonr/lilygo-tdisplays3-esphome
> 3. Copying the example.yaml and the tdisplays3 folder to my esphome dir.

Also try plugging the ESP32 board directly into a USB port on the PC that is running Home Assistant.

njbuch, 2023-04-05:
Good news, the @patfelst tips worked! I am not sure, but pulling the dev-branch from esphome might be the trick - instead of the going with the esphome-latest release version?

I will start experimenting with some larger display designs and some sensors. 

Thank you all the for the assistance! :)

closed 2023-04-05 by njbuch