landonr/lilygo-tdisplays3-esphome #13: implement touch

sbur83, 2023-03-01:
Anyone know will the tdiasplay-s3-touch variant work with this component?

guillempages, 2023-03-02:
@sbur83 I also have the touch variant and have implemented a very basic touch component. It's working for me, but I still need to debug it a bit before pushing it somewhere.

I've created a PR (#10) to this repository, to allow the display (without touch) to be used as an external component, with the standard ESPHome display component configuration; this allows using images, pages and all the fancy stuff.
If @landonr merges this PR, I can submit another one for the touch component.

guillempages, 2023-03-02:
@sbur83 despite my previous comment, I've created a branch on my fork of landonr's repository with my initial touch implementation. I haven't created the PR yet, because as I said, I'm still not 100% happy with the code.
But if you want to test it, you can check the `example-touch.yaml` in `github://guillempages/lilygo-tdisplays3-esphome` in the _add_touch_ branch.

In order to get it working for you, you will need to replace "landonr" with "guillempages" in the github address of the `external-components`, and add "@add_touch" after it. Like:

```
external_components:
  - source: github://guillempages/lilygo-tdisplays3-esphome@add_touch
    components: [tdisplays3]
```

afterwards you can use the standard display and touchscreen components from esphome:
```
display:
  - platform: tdisplays3

touchscreen:
  - platform: tdisplays3
    interrupt_pin: 16
    reset_pin: 21
```

Hope this helps!

sbur83, 2023-03-04:
Legend, I'll test it out

sbur83, 2023-03-04:
@guillempages thanks dude, I tried and it isn't working right. The display is working with ESPHome display component but when it starts the logs say found no i2c devices, strange thing is when I touch the screen it is logging - Failed to communicate!
I have this in the .yaml:
i2c:
 sda: GPIO17
 scl: GPIO18
 scan: true
 
touchscreen:
  - platform: tdisplays3
    interrupt_pin: 16
    reset_pin: 21
```
[23:09:20][C][i2c.arduino:052]: I2C Bus:
[23:09:20][C][i2c.arduino:053]:   SDA Pin: GPIO17
[23:09:20][C][i2c.arduino:054]:   SCL Pin: GPIO18
[23:09:20][C][i2c.arduino:055]:   Frequency: 50000 Hz
[23:09:20][C][i2c.arduino:058]:   Recovery: bus successfully recovered
[23:09:20][I][i2c.arduino:068]: Results from i2c bus scan:
[23:09:20][I][i2c.arduino:070]: Found no i2c devices!
[23:09:20][C][switch.gpio:076]: GPIO Switch 'Backlight'
[23:09:20][C][switch.gpio:099]:   Restore Mode: restore defaults to ON
[23:09:20][C][switch.gpio:031]:   Pin: GPIO38
[23:09:20][C][homeassistant.time:010]: Home Assistant Time:
[23:09:20][C][homeassistant.time:011]:   Timezone: 'AEST-10AEDT,M10.1.0,M4.1.0/3'
[23:09:20][C][lilygo_tdisplay_s3.touchscreen:125]: Lilygo T-Display S3 Touchscreen:
[23:09:20][C][lilygo_tdisplay_s3.touchscreen:126]:   Address: 0x15
[23:09:20][C][lilygo_tdisplay_s3.touchscreen:127]:   Interrupt Pin: GPIO16
[23:09:20][C][lilygo_tdisplay_s3.touchscreen:128]:   Reset Pin: GPIO21
[23:09:20][C][lilygo_tdisplay_s3.touchscreen:129]: Rotation: 270
[23:09:20][C][mdns:108]: mDNS:
[23:09:20][C][mdns:109]:   Hostname: s3
[23:09:20][C][ota:093]: Over-The-Air Updates:
[23:09:20][C][ota:094]:   Address: s3.local:3232
[23:09:20][C][ota:097]:   Using Password.
[23:09:20][W][ota:103]: Last Boot was an unhandled reset, will proceed to safe mode in 8 restarts
[23:09:20][C][api:138]: API Server:
[23:09:20][C][api:139]:   Address: s3.local:6053
[23:09:20][C][api:141]:   Using noise encryption: YES
[23:09:20][C][wifi_info:009]: WifiInfo IPAddress 'IP Address'
[23:09:20][C][wifi_info:009]:   Icon: 'mdi:fingerprint'
[23:09:36][E][lilygo_tdisplay_s3.touchscreen:069]: Failed to communicate!
[23:09:36][E][lilygo_tdisplay_s3.touchscreen:069]: Failed to communicate!
[23:09:36][E][lilygo_tdisplay_s3.touchscreen:069]: Failed to communicate!
[23:09:36][E][lilygo_tdisplay_s3.touchscreen:069]: Failed to communicate!
[23:09:36][E][lilygo_tdisplay_s3.touchscreen:069]: Failed to communicate!
[23:09:36][E][lilygo_tdisplay_s3.touchscreen:069]: Failed to communicate!
[23:09:36][E][lilygo_tdisplay_s3.touchscreen:069]: Failed to communicate!
```
Do I need to declare the i2c address somewhere?

guillempages, 2023-03-04:
Hi @sbur83 , thanks for testing and for the feedback. I don't really know why it isn't working for you; I'll have to look deeper into the issue.

The touchscreen component has an "address" parameter where you can override the default I2C address (0x15}; you might want to test that.

You are getting some Los when you touch the display, because again this is triggered by the interrupt, and not by any I2C communication. 

Another possibility could be, that the touch controller on your display is a different model than mine; on the example code from logo there are two different code paths, and I only implemented the one that worked for me. Unfortunately, I have no idea how to find out which controller is being used.

I do have an additional switch on my yaml file that i cooped from somewhere else; I don't think it will make a difference, but you might try with
```switch:
  - platform: gpio
    pin:
      number: GPIO15
      mode:
        output: True
    name: "Power On"
```
And force that to true.
Also, i assume so, but are you sure that you are using the right pond for the I2C configuration?
For me scl is 17 and SDA is 18.

sbur83, 2023-03-05:
@guillempages Happy to test. Swapping i2c pind stopped getting the failed to connect log and nothing happened at all. It is the CST816 Chip for the touch screen I believe, I will try confirm the address but strange the log does not find it on scan - I might make a blank/ bare minimum .yaml with i2c scan and try later.

guillempages, 2023-03-05:
@do you have a binary sensor with the "touchscreen" platform?
AFAIR there are no logs when touching, if it works; or at least only debug Lisa, which are normally not shown.

You could also try adding a lambda in the "on_touch" configuration on the touchscreen part, and log "it.x" and "it.y" there. There is an example for that in the documentation for other touchscreens. (I'm writing from my phone ATM, I cannot confirm exactly which one, I think etk_something)

landonr, 2023-03-06:
sorry I missed this. I merged the PR, let me know how it goes

sbur83, 2023-03-06:
@guillempages, thanks, you and landonr rock!
You are right; I had the i2c data and clock pins mixed up and I hadn't declared the address. 
The touch example you provided wouldn't compile until I made a slight change: I removed fullstop and added quotation marks in ESP_LOGI line of lambda eg.
```
touchscreen:
  - platform: tdisplays3
    interrupt_pin: 16
    reset_pin: 21
    address: 0x15
    id: my_touchscreen
    on_touch:
      - lambda: |-
         ESP_LOGI("tdisplays3-touch", "x:%d, y:%d",
           id(my_touchscreen).x,
           id(my_touchscreen).y
         );
      - switch.toggle: backlight
```

guillempages, 2023-03-10:
@sbur83 sorry about the typo. Fixed that. 
Once you got it compiling; did touch work for you, or still didn't?

sbur83, 2023-03-10:
@guillempages working,  thanks. The button area logs - x: 65496, y: 85 and the max number is 2000. Other than that it's working well. The refresh is a bit annoying so I have been trying different ways to refresh only as needed

sbur83, 2023-03-10:
[image]
[image]

guillempages, 2023-03-11:
@sbur83 I have fixed the flickering in a separate commit (#14); the touch implementation branch on my repo is based on that, so if you pull the latest code, you should be good to go.

I found out the reason for the 65496 coordinate: The button is actually left of the screen, which means it has negative coordinates. I was using unsigned integers, so it was wrapped around. I've fixed it to use (signed) integers instead, but it doesn't help much: ESPHome requires touch coordinates to be between 0 and 2000.

The reason I had not seen this before was because I am using a 90° rotation, whereas the example I pushed (and that you are using) uses 270° :-D

I've added the possibility of adding `x_offset` (and `y_offset`) to the touchscreen configuration to be able to use that; I've also updated the `example-with-touch.yaml` file

**Warning:** I've removed the `reset_pin` from the configuration since it was not being used at all, so your current yaml file will not compile unless you remove that from your touchscreen config as well.

Let me know if it works now for you, and then I'll create the PR

sbur83, 2023-03-12:
@guillempages Smashing it: the refresh is perfect now; x_offset: 40 works fine and thanks for the heads up on reset_pin :)

guillempages, 2023-03-12:
Ok, I've done the PR then now.

guillempages, 2023-06-25:
@landonr since touch has been implemented and merged, you might want to close this issue ;-)

closed 2023-06-27 by landonr