landonr/lilygo-tdisplays3-esphome #22: Ili9488 cannot be full screen

shangjiazi, 2023-04-11:
I have an esp32 and ili9488 connected in parallel, which have successfully lit up with your code, but did not fill the entire screen. When I tried to modify the resolution and chip model, it went black. Requesting help, what file should I use to make the screen full and display? thank you

bradmck, 2023-04-12:
I'm not sure if this will help you, but you can try the code I talk about in this post: [Issue 15](https://github.com/landonr/lilygo-tdisplays3-esphome/issues/15#issuecomment-1489488055)

You'd want to use the user_build_flags option and then specify all of the parameters needed for your display.

shangjiazi, 2023-04-15:
[image]
I have found the problem and successfully displayed it by replacing the original "spr" with "tft" at the marked position. I am a novice, may I ask what the problem is?

patfelst, 2023-04-15:
I think you need to add `expose_tft_objects: true` like this:
``` yamml
display:
  - platform: tdisplays3
    id: disp
    update_interval: 1s
    rotation: 270
    expose_tft_objects: true
```

patfelst, 2023-04-15:
Oh sorry, that only works on @bradmck's branch:
https://github.com/bradmck/lilygo-tdisplays3-esphome/tree/OptionToExposeTFTObjects

guillempages, 2023-04-15:
@shangjiazi I don't have an ILI9488 display to test, and I'm not sure exactly how it works. Maybe it doesn't have sprite support, and that's why the "spr" calls do not work?

bradmck, 2023-04-15:
I'm surprised the current version of this repo works at all since it's using hard-coded driver, pins, etc.  I suggested my branch just since it lets you custom define all of the device settings and therefore may just work with any display supported by the current TFT_eSPI.  I also don't have one of these and am not sure why the spr calls don't work.  I'm curious though, how's the refresh drawing directly to the TFT?  Does it flash at all?

And did you try using the ST7789 driver with 320 * 480?

patfelst, 2023-04-15:
Did you have the `USER_SETUP_ID` set to 146 as indicated ([here](https://github.com/Bodmer/TFT_eSPI/blob/master/User_Setups/Setup70d_ILI9488_S3_Parallel.h))? In your code above you show it set to `1`.

Also on the TFT_eSPI github page it says:
> 5. Support for the ESP32-S2, ESP32-S3 and ESP32-C3 has been added (DMA only on ESP32 S3 at the moment).

I don't know if DMA works with ESPHome.

shangjiazi, 2023-04-16:
Why TFT_ WIDTH=170 can only be 170, "sprite" can only work properly, I don't know where the error is? Very distressed

guillempages, 2023-04-16:
It might be a memory issue; if the width is "too big", it needs to much
memory to store the sprites, and this it disables it. I don't know, that's
just a guess though.

shangjiazi, 2023-04-16:
@guillempages Another interesting thing to discover is that if you still fill in the size of "ST7789" after "spr. createSite" in "tdisplay_s3. h", it will still display normally. Do you need to modify it somewhere to make it full screen?
[image]
[image]

shangjiazi, 2023-04-16:
@guillempages I don't think it's a memory issue. If I change it to 171, it will also fail. Can only be 170. Among these programs, where else is there a fixed area related to "st7789" that needs to be modified?

bradmck, 2023-04-16:
Have you used this board/display with any basic Arduino/PlatformIO sketches (with TFT_eSPI)?  On your question of where the dimensions are set....  If you are using my Beta branch there's only one spot that I set the width and height, and that's only if you don't specify `user_build_flags:`.  That's in the display.py file:
```
        cg.add_build_flag("-DTFT_WIDTH=170")
        cg.add_build_flag("-DTFT_HEIGHT=320")
```

shangjiazi, 2023-04-18:
@guillempages You are right, asking the spirit to create a 320 * 480 does require a lot of memory, so I can only add "spr. setColorDepth (1);" to reduce memory usage. It successfully displayed. Thank you for your help. The success of ILI9488 means that this great library will support many TFT parallel port screens

bradmck, 2023-04-19:
@shangjiazi - I'm working to add a `color_depth:` option to my Beta branch but I'm a bit of an idiot with this stuff so it's not quite working yet.  :)  I'll reply here when that's ready if you're interested in testing.  My beta branch is the one that also allows specifying `user_build_flags` for the TFT_eSPI library.

bradmck, 2023-04-20:
My [Beta branch](https://github.com/bradmck/lilygo-tdisplays3-esphome/tree/Beta) now supports a `color_depth:` option.  Valid choices are 1, 4, 8, and 16.  Default is 8.  I've done very little testing but it _seems_ to work.

shangjiazi, 2023-04-21:
@bradmck It has been tested and it has been successful. excellent. Suggest renaming your "Beta branch" to make it more widely known. It does not fully support "tdisplays3" and "ST7789", and all "tft_espi" serial ports are supported

bradmck, 2023-04-21:
@shangjiazi  - I'm glad that worked for you but I'm not quite ready with that branch.  In _theory_ since you can specify your own settings for TFT_eSPI it's quite possible that it will work with any display supported by that library.  But I don't have any other displays to test with at this point.  I'm not sure what you mean by your reference to "serial ports are supported"?  It technically _should_ work for both serial and parallel displays.

shangjiazi, 2023-05-04:
Nice to have a conversation with you again. Your code has indeed helped me a lot and has become my pleasure. I have made modifications to my code. Due to memory issues, you have added the "color_depth:" setting to the original code. I have made new progress again because memory cannot create 320 * 480 "eSprines", but two 320 * 240 "eSprines" can be created. It can achieve 8-bit 'color_depth', and we only need to_ display_ S3. h Create two eSpites and let it determine which "eSpite" to push to.

I am using translation software to communicate with you, hoping to arouse everyone's interest.

shangjiazi, 2023-05-04:
@bradmck Nice to have a conversation with you again. Your code has indeed helped me a lot and has become my pleasure. I have made modifications to my code. Due to memory issues, you have added the "color_depth:" setting to the original code. I have made new progress again because memory cannot create 320 * 480 "eSprines", but two 320 * 240 "eSprines" can be created. It can achieve 8-bit 'color_depth', and we only need to_ display_ S3. h Create two eSpites and let it determine which "eSpite" to push to.

I am using translation software to communicate with you, hoping to arouse everyone's interest.

clowrey, 2023-05-04:
I'm curious what display board do you have there? Or is it something custom?