landonr/lilygo-tdisplays3-esphome #24: Screen turns on but doesn't display any content

bvarick, 2023-06-04:
I can't figure out how to get the display to show anything on my T-Display-S3 (non-touch). I've tried the example yaml here and it gives me this error when compiling:

`
...
Compiling .pioenvs/s3/libe96/TFT_eSPI/TFT_eSPI.cpp.o
In file included from .piolibdeps/s3/TFT_eSPI/TFT_eSPI.h:96,
                 from .piolibdeps/s3/TFT_eSPI/TFT_eSPI.cpp:16:
.piolibdeps/s3/TFT_eSPI/Processors/TFT_eSPI_ESP32_S3.h:110:3: warning: #warning >>>>------>> DMA is not supported in parallel mode [-Wcpp]
  #warning >>>>------>> DMA is not supported in parallel mode
   ^~~~~~~
In file included from .piolibdeps/s3/TFT_eSPI/TFT_eSPI.cpp:16:
.piolibdeps/s3/TFT_eSPI/TFT_eSPI.h:970:8: warning: #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available! [-Wcpp]
       #warning >>>>------>> TOUCH_CS pin not defined, TFT_eSPI touch functions will not be available!
        ^~~~~~~
...
`

and then the screen turns on, but stays blank. The touch error makes sense as its not a touch screen. 

Any suggestions?

guillempages, 2023-06-07:
The touchscreen lines are not an error, just a warning. I do have the touchscreen variant and I also get those 😄

IIRC, I had the same problem after flashing, and thought it was an error in the component... The solution was easy in my case: unplug USB/per source and plug it again; after that it did show the image normally (also after ota updates).

bvarick, 2023-06-07:
I appreciate the response and suggestion!
I tried unplugging and replugging the USB after installing and still no content on the display.

bvarick, 2023-06-08:
I figured it out:
I had plugged in a temp/humidity sensor into GPIO pins 1&2, and 5v. I switched them to GPIo 43&44, and 3v, and it all works great. Which pins should I avoid?

closed 2023-06-19 by bvarick