landonr/lilygo-tdisplays3-esphome #32: Touch example resets/black screen

webworxshop, 2023-07-02:
Hello,

I'm having trouble using the touch example. As a precursor, I tried the LCD only example, which works perfectly.

When the board powers up on the touch example, I get the following in the logs several times (and nothing else):

```
WARNING Decoded 0x42095aba: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:839
[12:21:49]SPIWP:0xee
[12:21:49]mode:DIO, clock div:1
[12:21:49]load:0x3fce3808,len:0x43c
[12:21:49]load:0x403c9700,len:0xbec
[12:21:49]load:0x403cc700,len:0x2a3c
[12:21:49]entry 0x403c98d8
[12:21:50]ESP-ROM:esp32s3-20210327
[12:21:50]Build:Mar 27 2021
[12:21:50]rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[12:21:50]Saved PC:0x42095aba
```

This looks to me like the board is resetting a few times, but this stops after a few cycles. The board then sits there with a blank screen and is offline in Home Assistant. The red LED flickers (as with the LCD only example).

If it's any help, my board has the text "T_Display_S3 22-8-2 V12" on the back. Perhaps a different/newer version?

Please let me know of any more info I can provide to get this working.

Thanks in advance,

Rob

webworxshop, 2023-07-02:
OK, I managed to make some progress, I needed to specify `deassert_rts_dtr: true` to get the logging working correctly.

The device seems to be boot looping a few times and then starts up and connects to the wifi:

```
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyACM0 with baud rate 115200
[18:20:07]ESP-ROM:esp32s3-20210327
[18:20:07]Build:Mar 27 2021
[18:20:07]rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
[18:20:07]Saved PC:0x40056f59
[18:20:07]SPIWP:0xee
[18:20:07]mode:DIO, clock div:1
[18:20:07]load:0x3fce3808,len:0x43c
[18:20:07]load:0x403c9700,len:0xbec
[18:20:07]load:0x403cc700,len:0x2a3c
[18:20:07]entry 0x403c98d8
[18:20:07][   180][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[18:20:07]ESP-ROM:esp32s3-20210327
[18:20:07]Build:Mar 27 2021
[18:20:07]rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[18:20:07]Saved PC:0x4209a642
WARNING Decoded 0x4209a642: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:839
[18:20:07]SPIWP:0xee
[18:20:07]mode:DIO, clock div:1
[18:20:07]load:0x3fce3808,len:0x43c
[18:20:07]load:0x403c9700,len:0xbec
[18:20:07]load:0x403cc700,len:0x2a3c
[18:20:07]entry 0x403c98d8
[18:20:08][   181][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[18:20:08]ESP-ROM:esp32s3-20210327
[18:20:08]Build:Mar 27 2021
[18:20:08]rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[18:20:08]Saved PC:0x4209a642
WARNING Decoded 0x4209a642: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:839
[18:20:08]SPIWP:0xee
[18:20:08]mode:DIO, clock div:1
[18:20:08]load:0x3fce3808,len:0x43c
[18:20:08]load:0x403c9700,len:0xbec
[18:20:08]load:0x403cc700,len:0x2a3c
[18:20:08]entry 0x403c98d8
[18:20:08][   181][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[18:20:08]ESP-ROM:esp32s3-20210327
[18:20:08]Build:Mar 27 2021
[18:20:08]rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[18:20:08]Saved PC:0x4209a642
WARNING Decoded 0x4209a642: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:839
[18:20:08]SPIWP:0xee
[18:20:08]mode:DIO, clock div:1
[18:20:08]load:0x3fce3808,len:0x43c
[18:20:08]load:0x403c9700,len:0xbec
[18:20:08]load:0x403cc700,len:0x2a3c
[18:20:08]entry 0x403c98d8
[18:20:08][   181][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[18:20:08]ESP-ROM:esp32s3-20210327
[18:20:08]Build:Mar 27 2021
[18:20:08]rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[18:20:08]Saved PC:0x4209a642
WARNING Decoded 0x4209a642: esp_pm_impl_waiti at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_pm/pm_impl.c:839
[18:20:08]SPIWP:0xee
[18:20:08]mode:DIO, clock div:1
[18:20:08]load:0x3fce3808,len:0x43c
[18:20:08]load:0x403c9700,len:0xbec
[18:20:08]load:0x403cc700,len:0x2a3c
[18:20:08]entry 0x403c98d8
[18:20:09][   181][I][esp32-hal-psram.c:96] psramInit(): PSRAM enabled
[18:20:09]ESP-ROM:esp32s3-20210327
[18:20:09]Build:Mar 27 2021
[18:20:09]rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[18:20:09]Saved PC:0x4209a642
[... 72 lines cut ...]
[18:20:17][  6544][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 1 - SCAN_DONE
[18:20:18][  7573][V][WiFiGeneric.cpp:353] _arduino_event_cb(): STA Connected: SSID: [redacted], BSSID: [redacted], Channel: 11, Auth: WPA2_PSK
[18:20:18][  7575][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
[18:20:18][  7585][V][WiFiGeneric.cpp:367] _arduino_event_cb(): STA Got New IP:10.0.2.23
[18:20:18][  7604][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
[18:20:18][  7608][D][WiFiGeneric.cpp:991] _eventCallback(): STA IP: 10.0.2.23, MASK: 255.255.255.0, GW: 10.0.2.1
```

The device is then, in fact, pingable on that IP:

```
ping 10.0.2.23
PING 10.0.2.23 (10.0.2.23) 56(84) bytes of data.
64 bytes from 10.0.2.23: icmp_seq=1 ttl=254 time=101 ms
64 bytes from 10.0.2.23: icmp_seq=2 ttl=254 time=123 ms
64 bytes from 10.0.2.23: icmp_seq=3 ttl=254 time=169 ms
^C
--- 10.0.2.23 ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 2003ms
rtt min/avg/max/mdev = 101.046/131.023/169.263/28.455 ms
```

So esphome is at least running (unexplained initial bootlooping aside). The screen is still blank however and no touch events are logged, so it looks like the display and touchscreen driver are conflicting somehow?

I did suspect that this may be a power issue so I plugged the board into a 4.8A USB power supply, which gave the same result (device pingable, no display, no touch events logged).

landonr, 2023-07-03:
thanks for posting your findings. I don't have a touch device but maybe @guillempages has an idea

guillempages, 2023-07-03:
I have the same text on the board (so I assume we have the same HW version), and I can also reproduce the issue with the latest master branch. Moving back to some older commit is working again, so this must be something introduced lately.

I'll try to find out what the problem might be and do a PR accordingly.

guillempages, 2023-07-03:
Looks like it is an issue with PS-RAM.
On the PR by @Fabian-Schmidt, he changed the tft and spr members to be pointers; allocated dynamically on `setup()`.
If I change them back to plain variables, it's working again for me.

Fabian, do you have the touch version or the non-touch version of the T-Display-S3? There might be a difference on how the psram is configured/used if touch is there.
At least, I haven't been able to use psram at all on my device.

guillempages, 2023-07-03:
nope, not psram. I managed to get psram working (with the extra platformio options that Fabian helpfully added to the examples), but with the dynamically allocated tft and spi variables it's still crashing.

guillempages, 2023-07-03:
PR #34 should fix the issue. @webworxshop, can you test if it fixes it for you as well?

Fabian-Schmidt, 2023-07-03:
I don't have a touchscreen. The change to allocate in `setup` was due to minimize allocation during object creation to avoid OTA issues. ESP Home likes to create as little as possible in object creation and moves most to setup method.

webworxshop, 2023-07-03:
@guillempages I can confirm that your PR is working for me, in that the display is now running.

For some reason I'm not getting anything logged on touch events using the lambda in the example, but maybe I need to poke around with the logging more to get that working.

Thanks for the awesome work!

guillempages, 2023-07-03:
> I don't have a touchscreen. The change to allocate in `setup` was due to minimize allocation during object creation to avoid OTA issues. ESP Home likes to create as little as possible in object creation and moves most to setup method.

That's a good goal. I just changed the allocation of the tft object, but left the sprite allocation in the setup method. Otherwise it is not working for me; I cannot explain why :-(

@webworxshop I'll need to look at the touch issue some other time; it's now time to go to bed ;-) Let me/us know if you manage to get it working though. I'm not logging anything on touch, but have setup some binary sensors with platform touchscreen, and they work like a charm for me.

guillempages, 2023-07-03:
I think I have an idea why is crashing when initiating in `setup`. IIRC the touchscreen component is getting the height and width from the display component, and this two methods are calling the `tft` pointer. So initializing it on setup is probably to late.

I'll have to do some checks tomorrow to confirm that.

circuit69, 2023-07-04:
I have the touch screen version and have been having these same issues. Working fine until I add the code for the touchscreen. Once this is working again I hope to incorporate the touch function into HA. I'm still new to all of this stuff but looking forward to learning how it can be used.

Thank you for all your work.

closed 2023-07-05 by guillempages

webworxshop, 2023-07-08:
@guillempages Sorry to post on a now closed issue, but I just wanted to confirm that I did get the touchscreen logging working using the esphome `logger.log` action.

Thanks again for all your help.

PR #34 (2023-07-05): Fix crashloop in touchscreen devices
The touchscreen component needs to access width and height before the setup method of the display is completed.
This was causing a nullptr access, and a crash.

Added some default values to be able to get width and height when the setup method has not been executed yet.

Fixes #32