landonr/lilygo-tdisplays3-esphome #35: Time does not sync with HA until interactive action

ecapuano, 2023-07-12:
First thing, this project is awesome -- I am a total noob to esp32 and I'd be lost with this new board without these examples. 

I have successfully installed the example w/touch template and everything works as advertised except one little thing... When the board boots, it displays the default (presumably null) time of 1969-12-31 18:00:00 ... It never pulls current time from HA on its own until I do something to interact with it like pulling logs via ESPHome (via HA). Once I begin pulling logs, it instantly refreshes to current time, but never until then.

Am I holding it wrong?

ecapuano, 2023-07-12:
If it matters, I did deviate from the installation instructions slightly by doing the install directly from HA with the board connected to the Pi via USB... So I never did the steps to pull the modern install and flash on my PC using web.esphome -- however, the install worked flawlessly despite.

The board functions fine, just doesn't automatically sync with HA time until  I force it to.

ecapuano, 2023-07-12:
Log output

```
INFO ESPHome 2023.6.5
INFO Reading configuration /config/esphome/rack.yaml...
INFO Detected timezone 'America/Chicago'
WARNING GPIO0 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from rack.local using esphome API
INFO Successfully connected to rack.local
[14:08:21][I][app:102]: ESPHome version 2023.6.5 compiled on Jul 12 2023, 13:46:47
[14:08:21][C][wifi:543]: WiFi:
[14:08:21][C][wifi:379]:   Local MAC: 34:85:18:A9:E5:E4
[14:08:21][C][wifi:380]:   SSID: [redacted]
[14:08:21][C][wifi:381]:   IP Address: 192.168.1.174
[14:08:21][C][wifi:383]:   BSSID: [redacted]
[14:08:21][C][wifi:384]:   Hostname: 'rack'
[14:08:21][C][wifi:386]:   Signal strength: -54 dB ▂▄▆█
[14:08:21][C][wifi:390]:   Channel: 6
[14:08:21][C][wifi:391]:   Subnet: 255.255.255.0
[14:08:21][C][wifi:392]:   Gateway: 192.168.1.1
[14:08:21][C][wifi:393]:   DNS1: 192.168.1.1
[14:08:21][C][wifi:394]:   DNS2: 0.0.0.0
[14:08:21][C][logger:301]: Logger:
[14:08:21][C][logger:302]:   Level: DEBUG
[14:08:21][C][logger:303]:   Log Baud Rate: 115200
[14:08:21][C][logger:305]:   Hardware UART: UART0
[14:08:21][C][i2c.arduino:053]: I2C Bus:
[14:08:21][C][i2c.arduino:054]:   SDA Pin: GPIO18
[14:08:21][C][i2c.arduino:055]:   SCL Pin: GPIO17
[14:08:21][C][i2c.arduino:056]:   Frequency: 50000 Hz
[14:08:21][C][i2c.arduino:059]:   Recovery: bus successfully recovered
[14:08:21][I][i2c.arduino:069]: Results from i2c bus scan:
[14:08:21][I][i2c.arduino:071]: Found no i2c devices!
[14:08:21][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Button 1'
[14:08:21][C][gpio.binary_sensor:016]:   Pin: GPIO0
[14:08:21][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'Button 2'
[14:08:21][C][gpio.binary_sensor:016]:   Pin: GPIO14
[14:08:21][C][ledc.output:160]: LEDC Output:
[14:08:21][C][ledc.output:161]:   Pin GPIO38
[14:08:21][C][ledc.output:162]:   LEDC Channel: 0
[14:08:21][C][ledc.output:163]:   PWM Frequency: 2000.0 Hz
[14:08:21][C][ledc.output:164]:   Bit depth: 14
[14:08:21][C][light:103]: Light 'Backlight'
[14:08:21][C][light:105]:   Default Transition Length: 1.0s
[14:08:21][C][light:106]:   Gamma Correct: 2.80
[14:08:21][C][homeassistant.time:010]: Home Assistant Time:
[14:08:21][C][homeassistant.time:011]:   Timezone: 'CST6CDT,M3.2.0,M11.1.0'
[14:08:21][C][TDisplayS3:023]: T-Display S3 (ST7789)
[14:08:21][C][TDisplayS3:023]:   Rotations: 270 °
[14:08:21][C][TDisplayS3:023]:   Dimensions: 320px x 170px
[14:08:21][C][TDisplayS3:024]:   Update Interval: 1.0s
[14:08:21][C][lilygo_tdisplay_s3.touchscreen:127]: Lilygo T-Display S3 Touchscreen:
[14:08:21][C][lilygo_tdisplay_s3.touchscreen:128]:   Address: 0x15
[14:08:21][C][lilygo_tdisplay_s3.touchscreen:129]:   Interrupt Pin: GPIO16
[14:08:21][C][lilygo_tdisplay_s3.touchscreen:130]:   Rotation: 270
[14:08:21][C][lilygo_tdisplay_s3.touchscreen:131]:   Offset: (40, 0)
[14:08:21][C][lilygo_tdisplay_s3.touchscreen:132]:   Firmware version: 0
[14:08:21][C][psram:020]: PSRAM:
[14:08:21][C][psram:021]:   Available: YES
[14:08:21][C][psram:024]:   Size: 8191 KB
[14:08:21][C][mdns:112]: mDNS:
[14:08:21][C][mdns:113]:   Hostname: rack
[14:08:21][C][ota:093]: Over-The-Air Updates:
[14:08:21][C][ota:094]:   Address: rack.local:3232
[14:08:21][C][ota:097]:   Using Password.
[14:08:21][C][api:138]: API Server:
[14:08:21][C][api:139]:   Address: rack.local:6053
[14:08:21][C][api:143]:   Using noise encryption: NO
[14:09:13][D][api:102]: Accepted 192.168.1.53
[14:09:13][W][api.connection:097]: 192.168.1.53: Reading failed: BAD_INDICATOR errno=11
```
The error that shows up at the bottom happens periodically, but I don't think it's related since the time updates immediately once logs start flowing.

ecapuano, 2023-07-12:
Figured it out, user error :) Had a mismatch or missing API encryption key. Thanks again for the awesome work.

closed 2023-07-12 by ecapuano

landonr, 2023-07-12:
glad you figured it out!