landonr/lilygo-tdisplays3-esphome #41: my previously working project fails to compile with fatal error: User_Setup_Select.h: No such file or directory

spitfire, 2023-10-19:
My previously working project fails to compile with:
```
In file included from src/esphome/components/tdisplays3/t_display_s3.h:4,
                 from src/esphome/components/tdisplays3/t_display_s3.cpp:1:
.piolibdeps/t-display-s3-01/TFT_eSPI/TFT_eSPI.h:69:10: fatal error: User_Setup_Select.h: No such file or directory
```

is this related to https://github.com/landonr/lilygo-tdisplays3-esphome/pull/40 ?

landonr, 2023-10-20:
definitely related. you can revert for now by updating the component to this

```
  - source: 
      type: git
      url: https://github.com/landonr/lilygo-tdisplays3-esphome/
      ref: sept2023
    components: [tdisplays3]
```

guillempages, 2023-10-20:
@spitfire 
Hm. I've just tried with the latest ESPHome sources and the latest version of this repository, and I could not reproduce the error...
Can you check at the top of your compile log, which version of TFT_eSPI library is being pulled on your setup? My compilation pulls:
> HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
> - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
> - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
>Dependency Graph
>|-- WiFi @ 2.0.0
>|-- ESPmDNS @ 2.0.0
>|-- Update @ 2.0.0
>|-- noise-c @ 0.1.4
>|-- Wire @ 2.0.0
>|-- WiFiClientSecure @ 2.0.0
>|-- HTTPClient @ 2.0.0
>|-- pngle @ 1.0.0
>|-- TFT_eSPI @ 2.5.33

spitfire, 2023-10-20:
Processing t-display-s3-01 (board: esp32-s3-devkitc-1; framework: arduino; platform: platformio/espressif32@5.4.0)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Removing unused dependencies...
HARDWARE: ESP32S3 240MHz, 320KB RAM, 8MB Flash
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
Dependency Graph
|-- AsyncTCP-esphome @ 2.0.1
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- Update @ 2.0.0
|-- ESPAsyncWebServer-esphome @ 3.1.0
|-- DNSServer @ 2.0.0
|-- ESPmDNS @ 2.0.0
|-- noise-c @ 0.1.4
|-- ArduinoJson @ 6.18.5
|-- TFT_eSPI @ 2.5.33

spitfire, 2023-10-20:
No, I just went back to the component with tag you suggested above.

spitfire, 2023-10-26:
Cleaned build files and it works for me with master branch again

closed 2023-10-26 by spitfire