landonr/lilygo-tdisplays3-esphome #43: example.yaml doesn't build anymore

lecrams, 2023-11-22:
the esphome configuration (example.yaml) doesn't build anymore with the esphome version 2023.11.2
the error I get is:

INFO ESPHome 2023.11.3
INFO Reading configuration /config/esphome/lilygo-display.yaml...
Failed config

esp32: None
  
  Please specify flash_size within esp32 configuration only.
  board: esp32-s3-devkitc-1
  variant: ESP32S3
  framework: 
    version: 2.0.5
    source: ~3.20005.0
    platform_version: platformio/espressif32@5.4.0
    type: arduino

Any idea what can cause this? Version 2023.11.1 did not have this

landonr, 2023-11-22:
fixed in #44

closed 2023-11-22 by landonr

WarC0zes, 2023-12-21:
duplicate https://github.com/landonr/lilygo-tdisplays3-esphome/issues/42
I didn't get any answer and I found it on my own. I gave the solution.

@landonr thanks for the fix.