landonr/lilygo-tdisplays3-esphome #47: Example with touch... not working?

inakiarredondo, 2023-12-10:
Has anyone managed to get the example-with-touch to work? (with Lilygo T-Display-S3 Touch version)
Logs show this message: "[E][lilygo_tdisplay_s3.touchscreen:069]: Failed to communicate!" (8 times in a row, each time I touch the screen)
Any advice? Thank you.

guillempages, 2023-12-13:
The base touchscreen component was changed, and all external touchscreen components need some reworking.

I've got it to compile, but the touch coordinates are completely bonkers. Will keep looking.

guillempages, 2023-12-16:
Not looking good... I've got it working if I don't set the display rotation, but as soon as I set rotation 90 or 270, the y coordinates get halved (i.e. from 0-90 instead of 0-170).

On the plus side of things, there is a PR on the official ESPHome repo (https://github.com/esphome/esphome/pull/5941) that adds touchscreen functionality for exactly this same touch controller. And it's a PR by someone who was involved on the touch/display discussions that are causing the problem now.

So there is a chance, that official support for this touch screen is coming into ESPHome core "soon"

inakiarredondo, 2023-12-18:
Thank you for your answer.
How did you do to make it working just don't setting the display rotation? Just removing that line doesn't make any change for me...

PS: Fingers crossed for the official support "soon".

guillempages, 2023-12-20:
To make the "t-display-s3 touchscreen" work, I had to do many changes in the code, that I haven't pushed yet. The PR I mentioned above works "out-of-the-box", if the display rotation is 0 (i.e. not set).
```
external_components:
  - source: github://pr#5941
    components: [cst816]

touchscreen:
  - platform: cst816
    display: disp
    interrupt_pin: 16
   #...
```

inakiarredondo, 2023-12-21:
Could you share the full code (for me to try it)?
Thanks!

guillempages, 2023-12-21:
@inakiarredondo I've created a PR (#50) that should fix the compilation issues with 2023.12

Whether this also fixes the "Failed to communicate" issue you originally mentioned, I cannot tell, since I'm not seeing this (nor was seeing it before). But it is probably worth a try (it is actually a must, if you upgraded to ESPHome 2023.12.0 or later)

inakiarredondo, 2023-12-23:
Now, once updated to ESPHome 2023 12.3, I get this warning 
`Couldn't find any component that can be used for 'display::Display'. Are you missing a hub declaration?`
when editing, highlighting tdisplays3 on this section:
```
touchscreen:
  - platform: tdisplays3
    interrupt_pin: 16
    address: 0x15
    ...
```

guillempages, 2023-12-23:
Yes, with 2023.12.0 there were some changes done to the touchscreen base component, that in turn required changes to the implementations.

That's what #50 should fix, but somehow it does not get through CI, so we cannot merge it.
I'm also afk for two weeks, so unfortunately I cannot look into that at the moment.

You might want to take a look at that PR; if I am not mistaken I wrote a comment on how to use that (you need to add `@<branch>` to the external component, but I can't tell you the exact name of the branch)

guillempages, 2024-01-09:
@inakiarredondo The needed PR has been merged now. Can you try without the `@branch` part, and see if it is now working for you? Then we could close this issue

inakiarredondo, 2024-01-17:
Sorry for the delay.

Yes, I've tried without the @branch part, with ESPHome 2023.12.5 and also with 2023.12.7 ...but the "Failed to communicate!" keeps arising on the logs.

Iñaki

raphbag, 2024-04-24:
Hello, I have the same problem with this error : "[lilygo_tdisplay_s3.touchscreen:082]: Failed to communicate!" when I touch the screen with ESPHome 2024.4.1.
Is there any news about this ?

raphbag, 2024-04-27:
I finally solved my problem.
I had just reversed the sda and scl pin which is not the right one on the doc image (and the right one on your example).