landonr/lilygo-tdisplays3-esphome #49: After the ESPhome 2023.12.0 update, the ESP reboots in a loop.

WarC0zes, 2023-12-21:
Hi,
the compile worked, but ESP rebooted in a loop. I restore to version 2023.11.6, it's good ESP works.
I did a "clean build files" and recompiled, it's the same. I can't see the logs because I can't connect to the device.

Anto79-ops, 2023-12-21:
when I connect to the device using a serial cable and use the ESPHome webpage, it works as it should and able to see the logs, with everything looking normal.

The moment I remove the cable and plug it into a non-data usb, it goes into the loop thing again.

guillempages, 2023-12-21:
I'm also seeing this issue. I've commented out all display and touchscreen related configurations, and the reboot loop is still happening, so this is not related to the external component in this repository, but a more generic issue with ESPHome.

I think there was some mention on discord about some serial hardware uart needing to be configured with the latest version, but I'm not sure about the details; I'll try to dig it up and test that.

guillempages, 2023-12-21:
I managed to track the issue to https://github.com/esphome/esphome/pull/4853
As a workaround, you can add this to your `logger` configuration, that way the device does not enter the reboot loop, but you also cannot read the logs via USB.

```
logger:
  hardware_uart: UART0
```

Not a real solution, but a workaround.

Anto79-ops, 2023-12-21:
hey thanks! I this could workd for me and I'll try it soon. I don't really need the logs via USB on this device.

By the way, for an official fix, is this something coming from this repo or ESPHome? thanks

guillempages, 2023-12-21:
An official fix should come from the official esphome repo.

You might want to see if there is already an open issue there and create one if it isn't. Unfortunately I'm not at the computer anymore to do that myself 🙁

Anto79-ops, 2023-12-22:
looks like the fix is coming in 12,2 any moment now....thanks!

WarC0zes, 2023-12-22:
> looks like the fix is coming in 12,2 any moment now....thanks!

ESP32-S3 and ESP-IDF don't play well with USB_CDC and need USB_SERIAL_JTAG https://github.com/esphome/esphome/pull/5929

good new,  waiting for the release.

WarC0zes, 2023-12-22:
Fix with ESPHome 2023.12.2.
No more loop problem.

closed 2023-12-22 by WarC0zes