landonr/lilygo-tdisplays3-esphome #58: Scroll long text

geminchi, 2024-01-19:
Hello, I am trying to display artist and song info from HA sensor, but the value is too long to fit the width of the display and be readable. is there any way to make the text scroll across on this display?

landonr, 2024-01-20:
I have this working in homeThing on the now playing screen for artist and song text. I'm not sure of any alternatives https://github.com/landonr/homeThing

geminchi, 2024-01-24:
@landonr, that is a great project, but a little bit above my skill level. Was trying to find a simple yaml code to help with text scroll, but looks like I will need to learn c++ to accomplish what I am trying to do ;)

patfelst, 2024-01-24:
Hi @geminchi, if ESPHome used and gave access to display sprites it would be easier, but I don't think that's an option. There was another branch in this repo where a guy implemented TFT_eSPI library and exposed sprites, but I don't he continued with it.

I think doing character by character text scroll on each display update would be relatively easy. @landonr can you point out the code in homething that does the scroll, I might try and adapt it to a simple bit of display lambda code for @geminchi .

landonr, 2024-01-30:
@patfelst @geminchi you can see how i do marquee for homeThing here,
https://github.com/landonr/homeThing/blob/60f52316f1fcbb2fb22b8f0ac2440936f4c93ad8/components/homeThingDisplayState/homeThingMenuTextHelpers.cpp#L43

let me know if you have questions!

landonr, 2024-07-01:
@patfelst

closed 2024-07-01 by landonr