landonr/lilygo-tdisplays3-esphome #59: External power solution

valeriotaccardo, 2024-01-23:
GPIO15 need to be set to HIGH during boot to permit an external power on 5V pin. Without it, ESP don't turn on. 
I've done this:

esphome:
  # ...
  on_boot:
    priority: -100.0
    # ...
    then:
      - switch.turn_off: power_en
  
switch:
  - platform: gpio
    pin: GPIO15
    id: power_en
    restore_mode: ALWAYS_ON

valeriotaccardo, 2024-01-23:
Excuse me! This is the correct code!
esphome:

...
on_boot:
priority: -100.0
# ...
then:
**- switch.turn_on: power_en**

switch:

platform: gpio
pin: GPIO15
id: power_en
restore_mode: ALWAYS_ON

mploetner, 2024-03-14:
Hi thank you. Basically what I was looking for but could you share your complet code please? Seems I am making a mistake with it. Here is what I use:

`esphome:
  name: s3
  friendly_name: ESP32-S3 T-Display
  platformio_options:
    board_build.f_flash: 80000000L
    board_build.partitions: default_16MB.csv
    board_build.arduino.memory_type: qio_opi

on_boot:
  priority: -100.0

then:
  - switch.turn_on: power_en

switch:

platform: gpio
pin: GPIO15
id: power_en
restore_mode: ALWAYS_ON

external_components:
  - source: github://landonr/lilygo-tdisplays3-esphome
    components: [tdisplays3]

esp32:
  board: esp32-s3-devkitc-1
  variant: esp32s3
  framework:
    type: arduino
  flash_size: 16MB

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "6ada29f6f41ce1685d29d406efd25fa4"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "S3 Fallback Hotspot"
    password: "zQ9tuPKIfFMu"

time:
  - platform: homeassistant
    id: ha_time

binary_sensor:
  - platform: gpio
    pin: 
      number: GPIO0
      inverted: true
    name: "Button 1"
  - platform: gpio
    pin: 
      number: GPIO14
      inverted: true
    name: "Button 2"

sensor:
  - platform: homeassistant
    name: Temperatur
    id: temp
    entity_id: sensor.temp_sensor_test_temperature

output:
  - platform: ledc
    pin: GPIO38
    id: gpio38
    frequency: 2000

light:
  - platform: monochromatic
    output: gpio38
    name: "Backlight"
    restore_mode: RESTORE_DEFAULT_ON`

valeriotaccardo, 2024-03-15:
So, can I use the fixed code?

mploetner, 2024-03-15:
your code is already working. I made a stupid mistake

valeriotaccardo, 2024-03-15:
Perfect... I will try it tomorrow

<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Privo
di virus.www.avg.com
<http://www.avg.com/email-signature?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2024-07-01 by landonr