landonr/lilygo-tdisplays3-esphome #60: Support is now in mainline for these displays!!

clowrey, 2024-03-12:
Also the work this guy is doing is awesome!! 
https://esphome.io/components/display/ili9xxx.html

we will soon have LVGL too looks like! 
https://github.com/clydebarrow/esphome/tree/75c5d5f19fe789053d47fa6874df0e03d5bf1015/esphome/components/lvgl
https://gist.github.com/clydebarrow/9d2cffd739cb844e9d6d5005fd29518d

patfelst, 2024-03-14:
which of the many ILI9xxx models do you use for the T-Display S3 AMOLED?

mploetner, 2024-03-14:
> which of the many ILI9xxx models do you use for the T-Display S3 AMOLED?

should be ST7789V

landonr, 2024-03-14:
I had trouble using the display in support. What's the SPI config?

fightforlife, 2025-01-31:
Are you sure? I think this only supports SPi based connections. But on the T Display S3 it is wired in parrellel mode.
Is there an example config to get it to work?

guillempages, 2025-04-05:
There is out-of-the-box support in upstream ESPHome for the T-Display-S3 AMOLED version, but not for the tft one AFAIK.

IIRC, there is a PR open that would add direct support, but I'm not sure about the status. A while ago @clydebarrow was working on a different PR to add support, but for some non-disclosed reasons dropped that.

I'm not sure if he has resumed the work on the new PR, or is it something unrelated.

clydebarrow, 2025-04-05:
They're all supported in the new driver in PR.
https://github.com/esphome/esphome/pull/8383