landonr/lilygo-tdisplays3-esphome #63: Touch x=0 y=0

vix66, 2024-04-18:
I'm having trouble with touch. Everything works fine if I compile and flash the device. At the first restart the touch works but the coordinates are x=0 and y=0. What can it be ?

vix66, 2024-04-20:
Am I the only one having this problem?

guillempages, 2024-04-21:
I have just tested, and I do get coordinates for the touch events (i.e. not 0,0).

My problem is, that if I use some rotation, the coordinates do not match the expected ones. I think this has already been going on for a while (since some refactoring was done on the upstream display/touch components), but unfortunately I haven't goten around to fixing that.

Nevertheless, for touch, there is now an official driver on core esphome, that should be used instead; the t-display-s3 specific touch controller is in the process of being deprecated.

@vix66 Could you try using `- platform: cst816` instead of `- platform: tdisplays3` for your touchscreen component?

vix66, 2024-04-23:
Thanks but it's the same. With a factory reset the touch starts working again, so I set that when the coordinates are x=0 and y=0 I do a reset. This happens at power on.