landonr/lilygo-tdisplays3-esphome #64: Dimmable?

dalanmiller, 2024-06-18:
Hi there 🙋🏻‍♂️, 

I don't quite understand this bit in the example yaml. 

```
output:
  - platform: ledc
    pin: GPIO38
    id: gpio38
    frequency: 2000

light:
  - platform: monochromatic
    output: gpio38
    name: "Backlight"
    restore_mode: RESTORE_DEFAULT_ON

# You can either setup the backlight as a switch like below or make it dimmable using a light output like above
# switch:
#   - platform: gpio
#     pin: GPIO38
#     name: "Backlight"
#     id: backlight
#     internal: true
#     restore_mode: RESTORE_DEFAULT_ON
```

How does this make the display dimmable? I can't seem to alter the display brightness whatsoever and I'm trying to lower the power consumption. 

This sources imply that it is both configurable and others have found success in doing so:

https://esphome.io/components/output/ledc.html
https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/74

But I don't see how to replicate the raw Arduino code mentioned below in esphome? 

https://github.com/Xinyuan-LilyGO/T-Display-S3/issues/74#issuecomment-1366378638

landonr, 2024-06-19:
I think the frequency should be 5000, instead of 2000. I'll try that and change it if it works for me.
Besides that, have you tried turning the backlight on and off again? You can connect to the ESP32 using home assistant, or the web server component to control the backlight.

otherwise i'm not sure how to help. Are you able to turn the backlight off and on?

dalanmiller, 2024-06-19:
This is what I'm seeing in Home Assistant at the moment. I can turn the backlight on and off but I can see that there's a fade out and in effect possible which also to me indicates that I _should_ be able to keep it dimmer than I'm defaulting to right now. 

Let me set the freq to `5000` and see if that changes anything, wouldn't I need to highlight to ESPHome that this is a setting that is continuous rather than binary off/on? 

https://github.com/landonr/lilygo-tdisplays3-esphome/assets/863286/3c5c5e65-0872-4010-ba16-867915190312

[image]

landonr, 2024-06-19:
try tapping and holding on the light name in that home assistant menu to get into brightness controls, or using a service call for turn_on with brightness provided

closed 2024-07-01 by landonr

dalanmiller, 2024-07-09:
> try tapping and holding on the light name in that home assistant menu to get into brightness controls, or using a service call for turn_on with brightness provided

This Home Assistant UI was entirely obvious to me but this worked exactly as @landonr described!